verilator/verilator #6172: Not Verilate Unwanted Top Module signal

jiwoop, 2025-07-10:
Is there a way to not verilate ports (both in/output) in top module? Other submodules in design use these ports, but when we run simulation on the verilated model we don't use those signals at all.

wsnyder, 2025-07-10:
Not directly. Make a wrapper module using verilog-mode or otherwise that stub or tie off the signals, then Verilate that.

jiwoop, 2025-07-10:
Thank you for your response. Let's say I had this module with two unused ports:
```
module real_top (
    input logic clk,
    input logic reset,
    input logic unused_in,
    output logic unused_out
);
    // Used by submodules, but not needed in C++
endmodule
```

Then, if I were to Verilate with this wrapper top module, would `unused_in` and `unused_out` either be optimized/eliminated?
```
module verilated_top (
    input logic clk,
    input logic reset
);

    logic unused_in = 1'b0;
    logic unused_out;

    real_top u_real (
        .clk(clk),
        .reset(reset),
        .unused_in(unused_in),
        .unused_out(unused_out)
    );

endmodule
```