verilator/verilator #6179: `public` module implies `/* verilator public */` for variables?

JiwooPark-arm, 2025-07-15:
It is stated in the doc that e.g., doing `/*verilator public_flat*/` for variables will not declare the containing module public. However, when we declare a module `public`, would that automatically make the variables within public?

JiwooPark-arm, 2025-07-15:
Also clarification. If I were to do
```
module A
// Parameter List
#(
  parameter WIDTH /*verilator public*/ = 2,
  parameter VIEWS /*verilator public*/ = 2
)
```
Then, in a Verilated file, they should remain as WIDTH and VIEWS instead of being converted to numbers right?

wsnyder, 2025-07-15:
>When we declare a module public, would that automatically make the variables within public

Yes.

>[Parameters] should remain as WIDTH and VIEWS instead of being converted to numbers right?

No, all references are numbers, but the parameter itself is available as a const identifier should you want to get the value from C++.

closed 2025-07-15 by wsnyder