verilator/verilator #6180: Support IEEE Std 1800-2017, section 27.5, Conditional generate constructs

qaztronic, 2025-07-15:
Thanks for taking the time to report this.

What would you like added/supported?

Please add support for IEEE Std 1800-2017, section 27.5, Conditional generate constructs.

What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.038 2025-07-08 rev v5.038

Can you attach an example that runs on other simulators?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

Here is the example from IEEE Std 1800-2017, section 27.5

module top;
  // -----------------------------------------------------------------------------
  parameter p = 1, q = 0;
  
  /* verilator lint_off UNUSEDSIGNAL */
  wire a = 0;
  wire b = 0;
  wire c = 0;
  /* verilator lint_on UNUSEDSIGNAL */
  
  //---------------------------------------------------------
  // Code to either generate a u1.g1 instance or no instance.
  // The u1.g1 instance of one of the following gates:
  // (and, or, xor, xnor) is generated if
  // {p,q} == {1,0}, {1,2}, {2,0}, {2,1}, {2,2}, {2, default}
  //---------------------------------------------------------
  
  /* verilator lint_off GENUNNAMED */
  if (p == 1)
    if (q == 0)
    begin : u1 // If p==1 and q==0, then instantiate
      reg foo;
      and g1(a, b, c); // AND with hierarchical name test.u1.g1
    end
    else if (q == 2)
    begin : u1 // If p==1 and q==2, then instantiate
      or g1(a, b, c); // OR with hierarchical name test.u1.g1
    end
    // "else" added to end "if (q == 2)" statement
    else ; // If p==1 and q!=0 or 2, then no instantiation
  else if (p == 2)
    case (q)
      0, 1, 2:
      begin : u1 // If p==2 and q==0,1, or 2, then instantiate
        xor g1(a, b, c); // XOR with hierarchical name test.u1.g1
      end
      default:
      begin : u1 // If p==2 and q!=0,1, or 2, then instantiate
        xnor g1(a, b, c); // XNOR with hierarchical name test.u1.g1
      end
    endcase
    /* verilator lint_on GENUNNAMED */

  // -----------------------------------------------------------------------------
  // --------------------------------------------------------------------
  bit clk;
  initial forever #(10ns/2) clk = ~clk;  
  
  // -----------------------------------------------------------------------------
  initial begin
    repeat(8) @(posedge clk);
    test.u1.foo <= 0;
    repeat(8) @(posedge clk);
    // test.u1.foo = 1;
    $finish;
  end

// -----------------------------------------------------------------------------
endmodule

Here is the error message:

verilator -sc --exe -x-assign fast --trace --assert --coverage --timing -f files.f ../../src/sc/sc_main.cpp
%Error: top.sv:80:9: Can't find definition of 'u1' in dotted scope/variable: 'top.u1'
   80 |     top.u1.foo <= 0;
      |         ^~
        ... Known scopes under 'top': <no instances found>
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.038 for more assistance.

May we assist you in trying to fix this in Verilator yourself?

qaztronic, 2025-07-15:
looks like it's working in Cadence Xcelium.

[image]

wsnyder, 2025-07-15:
Verilator supports `generate if`, it just hasn't been updated to support the IEEE naming that came in a later standard. See #4769. Pulls to fix that are welcome, please comment there.

closed 2025-07-15 by wsnyder