verilator/verilator #6189: Incorrect signal range order for last dimension when reading range via VPI

iburaky2, 2025-07-17:
Version:
Verilator 5.036 2025-04-27 rev v5.036

OS:
AlmaLinux 8.10

For a signal:
`logic[26:72][52:73][123:33][3:42] multidim_signal`

when reading the ranges via VPI:
```
vpiHandle iter_h = vpi_iterate(vpiRange, vh);
while (vpiHandle range_h = vpi_scan(iter_h)) {
    vpi_get_value(vpi_handle(vpiLeftRange, range_h), &value);
    left = value.value.integer;
    vpi_get_value(vpi_handle(vpiRightRange, range_h), &value);
    right = value.value.integer;
    std::cout << "Dim " << i << ": [" << left << ":" << right << "]" << std::endl;
}
```

The last dimension is converted to descending order, i.e. `[42:3]` is read instead of `[3:42]` regardless of how the signal is defined in Verilog.

This only happens for the last dimension, the rest are read however they are defined.

I'll upload an example in a second.

iburaky2, 2025-07-17:
https://github.com/iburaky2/verilator-issue-6189

wsnyder, 2025-07-17:
Did you confirm other simulators get this correct?  Might you submit a pull request to fix this (V3Emit{something}.cpp likely needs to use varp->range.left() and .right() instead of lo() and hi()).

PR #6200 (2025-07-20): Fix VPI signal range order (#6189)
This PR fixes an issue where the final dimension of a signal is converted to descending range during verilation regardless of how it is defined in Verilog.

Also adds a test_regress test for this case.

Closes #6189.
closed 2025-07-20 by wsnyder