verilator/verilator #6190: Smoke test fails on macOS due to missing distro module

expsg-emilgoh, 2025-07-17:
Hello,

I'm encountering an issue when running `make test` on macOS using `v5.036` and `v5.038`. I did not have this problem for `v5.034`.

The following is the error message:
```
Traceback (most recent call last):
  File "/verilator/test_regress/./driver.py", line 27, in <module>
    import distro
ModuleNotFoundError: No module named 'distro'
```

It seems like `distro`  is not installed on macOS by default which causes the smoke test to fail.

My workaround:
Commented out the following:
- https://github.com/verilator/verilator/blob/7f1011e5f7d194e62e25ceae090d935baf223d0c/test_regress/driver.py#L27
- https://github.com/verilator/verilator/blob/7f1011e5f7d194e62e25ceae090d935baf223d0c/test_regress/driver.py#L1264-L1269

System information:
- macOS version: macOS Sequoia 15.5
- Python version: Python 3.13.0
- Versions tested: `v5.036` and `v5.038`

paul-demo, 2025-07-26:
Yes you have to pip install distro. I’ve seen this too on MacOS.

wsnyder, 2025-07-27:
Note `python3-distro` is listed as needed in install.rst.  Assuming that's enough, otherwise pull requested.