verilator/verilator #6198: Optimizing 2 ** X into left-shift (missed optimization)

MaxWipfli, 2025-07-18:
**What would you like added/supported?**

It would be nice if Verilator could optimized `2 ** x` (where `x` is unsigned) into `1 << x`. This currently works if the base is constant and `unsigned` (e.g. `32'd2`), but the optimization is not applied if the base is declared as signed.
This construct is quite common (even though one could argue people should just use an unsigned base for correctness).

```systemverilog
int unsigned x;

// optimized into 1 << X (left-shift)
assign opt1 = 32'd2 ** X;
assign opt2 = $unsigned(2) ** X;

// optimization missed
assign noopt1 = 2 ** X;
```

**Note about Signedness**

I am not sure about the required output type. As far as I can see, Verilator produces a signed output, but I am unsure whether this is correct. There is this comment in the source code which I don't understand fully:
https://github.com/verilator/verilator/blob/9b99d9697fd631d46b7bdaf0713a4b9d82348138/src/V3Width.cpp#L1499-L1501

In any case, this optimization can still work even if the output needs to be signed (I think), by converting the output back to a signed number after shifting.

There is also [this commit](https://github.com/verilator/verilator/commit/55ad95060996d3398ef8eee883f5ec71d935e731) which removed this optimization, and I assume there was some thought behind that. So we need to be careful to ensure a correct output type.

**What 'verilator --version' are you using?  Did you try it with the git master version?**

`Verilator 5.039 devel rev v5.038-1-g8ba7cec15` (very recent)

**May we assist you in trying to fix this in Verilator yourself?**

I am trying to prepare a patch for this.

MaxWipfli, 2025-07-20:
Thanks for getting back to me, and sorry for the overloaded report. You are right that the commit I linked is primarily concerned with the special case where the exponent was zero. Sorry about that.

As explained in the first part of my report, I am aware this optimization already exists (see the snippet you provide). However, this only applies to the `AstPow` node, in which both base and exponent are unsigned.

What I was looking for is the case where the base is a signed constant (which you get by simply writing `2` instead of the unsigned `32'd2`).

Here is a self-contained test case:

```systemverilog
// test.sv
module test (
  input  int unsigned exponent,
  output int unsigned result_a,
  output int unsigned result_b
);
  assign result_a = 32'd2 ** exponent; // unsigned base
  assign result_b =     2 ** exponent; // signed base
endmodule
```
```
$ verilator --version
Verilator 5.039 devel rev v5.038-46-gf535a73ea
$ verilator --cc test.sv
...
$ grep -rE "SHIFT|POW" obj_dir/
obj_dir/Vtest___024root__DepSet_hc07518e4__0.cpp:    vlSelfRef.result_a = VL_SHIFTL_III(32,32,32, (IData)(1U), vlSelfRef.exponent);
obj_dir/Vtest___024root__DepSet_hc07518e4__0.cpp:    vlSelfRef.result_b = VL_POWSS_III(32,32,32, (IData)(2U), vlSelfRef.exponent, 1,0);
```

As you can see, the case where the constant base is unsigned is correctly optimized (`AstPow`), but the case where the base is signed (`AstPowSU`) is missed.

wsnyder, 2025-07-20:
Probably need `operandIsTwo` to also specially know that the MSB when signed (e.g. 4'sb1000) isn't a power of two and return false.

wsnyder, 2025-07-20:
Thanks for finding and fixing.