verilator/verilator #6199: Incorrect handling of x in if blocks (if (x) else ) ?

toby-bro, 2025-07-19:
Hello,

I know this is related to #3645, but I thought this is not that much about x propagation than about the "truthiness" of `x`. And when I tested this with other simulators (on [edaplayground](https://edaplayground.com/x/Jrwi)) I got the output of 1, which is what I would have expected from hand execution.

```sv
module minimal_repro (
    input logic clk,
    output logic out2
);
    logic condition;
    logic q1 = 1'bx;

    always @(posedge clk) begin
        q1 <= 0;
    end
    assign condition = ~q1;

    always @(posedge clk) begin
        if (condition) begin

        end else begin
            out2 <= 1'b1;
        end
    end
endmodule

module top;
    logic clk;
    logic out2;

    minimal_repro uut (
        .clk(clk),
        .out2(out2)
    );

    initial begin
        clk = 0;
        #10;
        repeat (5) begin
            clk = 0;
            #5;
            clk = 1;
            #5;
            $write("out2: %b\n", out2);
        end
        $finish;
    end
endmodule
```

Simulated with `verilator --binary --exe --build -Mdir obj_dir --timing -sv top.sv -O3 --quiet && ./obj_dir/Vtop`

The verilator version is 
`Verilator 5.038 2025-07-08 rev v5.038` 

I am on linux (debian 12)

> May we assist you in trying to fix this in Verilator yourself?
I am not sure I am capable of solving this. 

I would have expected: 
| cycle | q1 | condition | out |
|---------|-----|---------------|-------|
| 0 | x | x | 1 |
| 1 | 0 | 1 | 1 |
| 2 | 0 | 1 | 1 |
| 3 | 0 | 1 | 1 |
| ... | 0 | 1 | 1 |

Because the if (condition) should be evaluated to false as x is not truthy, so out should be assigned the value 1 at first clock cycle, for the others it's value should not change. But I get `out2 = 0` at each step so I think this is a bug. (If it is not I am sorry to have bothered you.)

Thank you for the awesome work on verilator.

wsnyder, 2025-07-19:
Are you sure you are using the latest Verilator version from sources here?

toby-bro, 2025-07-21:
Hello, I used the latest release 5.038 and just checked with origin/master and still get this output which I would have thought incorrect
```txt
out2: 0
out2: 0
out2: 0
out2: 0
out2: 0
- top.sv:41: Verilog $finish
- S i m u l a t i o n   R e p o r t: Verilator 5.039 devel
- Verilator: $finish at 60ps; walltime 0.001 s; speed 35.933 ns/s
- Verilator: cpu 0.002 s on 1 threads; alloced 120 MB
```

wsnyder, 2025-07-23:
When I run with the latest (from here, not release) I get all ones.

toby-bro, 2025-07-25:
This is weird.
I just tried again on another computer from a clean install on debian 12 from cloning master and recompiling from source and got all zeros.

closed 2025-07-25 by wsnyder