verilator/verilator #6211: Bug with DFG same component

MrPoloGit, 2025-07-21:
Hello,

I'm encountering a bug while trying to run Yosys synthesis and then using Verilator. This example works with Verilator versions before 2025-07-15, but afterwards it's broken.

Is it possibly that https://github.com/verilator/verilator/pull/6176 is the problem? @gezalore 

Versions:
```bash
Linux Mint 21.2
Verilator 5.039 devel rev v5.038-26-gdb6b17fdb
```

Minimal Example:
```verilog
module mul (input [8:0] A, input [16:0] B, output [25:0] Y);
  assign Y = $signed(A) * $signed(B);
endmodule

module A;
  wire [26:0] C;
  wire [26:0] D;
  wire [8:0]  E;

  mul mul (
    .A(9'd10),
    .B(17'h0cccd),
    .Y({ C[26], C[9:0], D[15:1] })
  );

  assign E = { C[8:0] };
  assign C[25:10] = {16{C[26]}};

endmodule
```

```bash
verilator --lint-only test.v --top A
```

Output:
```bash
%Error: Internal Error: test.v:8:15: ../V3DfgDecomposition.cpp:354: Driver of DfgVertexVar must be in the same component
    8 |   wire [26:0] D;
      |               ^
                        ... See the manual at https://verilator.org/verilator_doc.html?v=5.039 for more assistance.
```

Is there any more information that would be useful?

gezalore, 2025-07-21:
Thanks, the small example is great. I will look into this tomorrow.

sifferman, 2025-07-22:
Thanks for your help, @gezalore!