verilator/verilator #6214: Super constructor calls with local variables doesn't work

igorosky, 2025-07-22:
## Example
```verilog
class Base;
  function new(int x);
  endfunction
endclass
class Derived extends Base;
  function new;
    int x = 1;
    super.new(x);
  endfunction
endclass
```
Above code snippet verilates but fails on ``C++`` compilation stage.

Error:
```
In file included from Vl__ALL.cpp:8:
Vl___024unit__03a__03aDerived__Vclpkg__DepSet_h3f0e46e6__0.cpp: In constructor ‘Vl___024unit__03a__03aDerived::Vl___024unit__03a__03aDerived(Vl__Syms*)’:
Vl___024unit__03a__03aDerived__Vclpkg__DepSet_h3f0e46e6__0.cpp:10:43: error: ‘x’ was not declared in this scope
   10 |     : Vl___024unit__03a__03aBase(vlSymsp, x) {
      |                                           ^
```

Command:
```bash
$VERILATOR --binary -O0 file.v
```
Important here is ``-O0``. However, this issue can be replicated without ``-O0`` with a bit more complex example.

## Cause
Verilator generates ``C++`` code presented below:
```C++
Vm___024unit__03a__03aDerived::Vm___024unit__03a__03aDerived(Vm__Syms* __restrict vlSymsp)
    : Vm___024unit__03a__03aBase(vlSymsp, x) {
    VL_DEBUG_IF(VL_DBG_MSGF("+          Vm___024unit__03a__03aDerived::new\n"); );
    // Init
    _ctor_var_reset(vlSymsp);
    // Body
    IData/*31:0*/ x;
    x = 0;
    x = 1U;
    ;
}
```

The problem is that `x` variable is referenced in the call of the base class constructor, but the declaration is in the body below.

## Environment:
### Verilator version
```bash
$VERILATOR --version
Verilator 5.039 devel rev UNKNOWN.REV
```
(built from this [commit](https://github.com/verilator/verilator/commit/98b8d43a4ae52dba9d42422e61c76c52be5d5c0d))

### **OS**
``Debian 12``
### **g++**
```bash
g++ --version
g++ (Debian 12.2.0-14+deb12u1) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

## Solutions
Here we want to propose two potential solutions to this issue - both changing the way objects in ``C++`` are constructed.

Solution will use the following code as an example (it contains more complex cases than the example above).
```verilog
class Base;
  function new(int x);
  endfunction
endclass
class Derived extends Base;
  int a;
  int b = 3;
  function new;
    int x = 1;
    super.new(x);
    $write("%d\n", x);
    b += 4;
    a = 7;
    $write("x: %d b: %d\n", x, y, b);
  endfunction
endclass
```

### Solution 1 - ``init`` method
First solution is to instantiate empty object first and then use ``init()`` method as a proper object setup.

Example ``C++`` code for example above:
```C++
#include <iostream>

struct VlSymsp {};

class Base {
public:
    Base(){}
    Base(VlSymsp vlSymsp, int o){ init(vlSymsp, o); }
    void init(VlSymsp vlSymsp, int o) { std::cout << "[DEBUG]: GOT: " << o << '\n'; }
    Base(const Base&) {std::cout << "[DEBUG]: COPY\n";}
    Base(Base&&) {std::cout << "[DEBUG]: MOVE\n";}
};

class Derived : Base {
    int a;
    int b = 3;
public:
    Derived(){ }
    Derived(VlSymsp vlSymsp) { init(vlSymsp); }
    void init(VlSymsp vlSymsp) {
        int x = 1;
        Base::init(vlSymsp, x);
        b += 4;
        a = 7;
        std::cout << "x: " << x << " b: " << b << '\n';
    }
};
```
It is simple, but I don't like the fact it instantiates empty objects and then fixes them.

### Solution 2 - static factory function
Using static factory function to allow calls before construction of base object and allow sharing variables between before and after super call.

Example ``C++`` code for above example:
```C++
#include <iostream>

class Base {
    struct SuperArgs {};
    Base(SuperArgs&&, int o) { std::cout << "[DEBUG]: GOT: " << o << '\n'; }
public:
    Base(const Base&) {std::cout << "[DEBUG]: COPY\n";}
    Base(Base&&) {std::cout << "[DEBUG]: MOVE\n";}
    static Base create(int o) {
        return Base(SuperArgs{}, o);
    }
};

class Derived : Base {
    int a;
    int b /* = 3*/;
    struct SuperArgs {
        int x;  // We have to add here every variable that is being declated before super
    };
    Derived(SuperArgs&& args): Base{Base::create(args.x)}, b(3) {
        std::cout << args.x << '\n';
        b += 4;
        a = 7;
        std::cout << "x: " << args.x << " b: " << b << '\n';
    }
public:
    static Derived create() {
        return Derived(SuperArgs{
            /*.x: */ 1,
        });
    }
};
```

It is much more complex and involves calling move constructor of the base class. However, it doesn't instantiate empty objects.

I believe that first one is much simpler and if there is no complex objects inside classes it can be used. However, if for some reason we do not want to create empty uninitialized objects then second solution satisfies us.

Which solution shall be implemented? Do you have any other ideas?

igorosky, 2025-07-30:
> init() doesn't work in all cases depending on the object types involved, and can also be slower, so I would prefer not to do that.

Could you provide an example of types for which approach with ``init()`` method doesn't work because I couldn't find any example?

wsnyder, 2025-07-30:
Any class which has constructor arguments can't use init(). Also init() likely will cause the underlying objects to be initialized twice (first in constructor then init) which in some cases can be optimized to one, but not always.  We should follow standard C++ guidance to avoid init unless no alternative, e.g. google coding recommendations and https://abseil.io/tips/42

gezalore, 2025-08-20:
For the record, the original case with the local variable is illegal. IEEE-1800 2023 section 8.15:

> A super.new call shall be the first statement executed in the constructor. This is because the superclass
shall be initialized before the current class and, if the user code does not provide an initialization, the
compiler shall insert a call to super.new automaticall

Also in 8.7:

> The new method of a derived class (see 8.13) shall first call its base class constructor
[super.new() as described in 8.15].

The second case using a base class method call as argument to `super` should also be illegal (the base class is not initialized at that point).

If you want to pass computed values to the superclass constructor, use a function or static method (same as C++). 

These two fail with your original syndrome, but they **kind of work** with `-fno-localize`, so it's a bug in V3Localize (don't localize variables involved in constructor arguments).

```systemverilog
module top;
  function static int f1;
    int n;
    return ++n;
  endfunction
  class Base;
    function new(int x);
    endfunction
  endclass
  class Derived extends Base;
    function new;
      super.new(f1());
    endfunction
  endclass
  Derived d = new;
endmodule
```

```systemverilog
module top;
  class Base;
    function new(int x);
    endfunction
    static function int f1;
      return 1;
    endfunction
  endclass
  class Derived extends Base;
    function new;
      super.new(f1());
    endfunction
  endclass
  Derived d = new;
endmodule
```

The reason I am saying they **kind of work** is because the generated C++ is actually a bit dodgy, the function output argument is inside the C++ derived class, and is strictly speaking uninitialized (in C++) at that point. I'm not sure if this matters but in this, `f1` will return garbage on Verilator, while it should return 32'hxxxx0fxx which in 2-state Verilator world should be 32'h00000f00. 

```systemverilog
module top;
  class Base;
    function new(int x);
    endfunction
    function automatic bit [31:0] f1;
      f1[15:8] = 8'hf;
    endfunction
  endclass
  class Derived extends Base;
    function new;
      super.new(f1());
    endfunction
  endclass
  Derived d = new;
endmodule
```

Actually, looking at the C++, which is just `f1__Vfuncrtn = (0xf00U | (0xffff00ffU & f1__Vfuncrtn));`, where `f1__Vfuncrtn` is a reference, this is broken anyways, as the return value should always be fully written. This should always return h00000f00 or h0000000f, but because of the reference output you will get h00000f0f once both branches execute once:

```
    function automatic bit [31:0] f1;
      if (a) 
        f1[15:8] = 8'hf;
      else
        f1[7:0] = 8'hf;
    endfunction
```

This should be a compile error:

```systemverilog
module top;
  class Base;
    function new(int x);
    endfunction
    function int f1;
      return 1;
    endfunction
  endclass
  class Derived extends Base;
    function new;
      super.new(f1()); // can't call f before initializing Base
    endfunction
  endclass
  Derived d = new;
endmodule
```

So in the short term:

- [ ] Fix V3Localize not to localize that variable
- [ ] Add error if super.new is not the first statement in the constructor
- [ ] Add error on any non-static method call in the constructor
- [ ] Write tests with all combinations of `function automatic` (proper function), `function static` (proper function), `static function` (static class method) in super call arguments - make sure those functions are not inlineable. 

In the long term we should covert all classes to structures and methods to functions internally, and emit them as C structs and functions. I don't see what we gain besides not having to do that work (which is a reason just not a strong one if we can do better).

gezalore, 2025-08-20:
Agree long time that's fine.

> there's still the issue of initializing superclasses with local vars because of how Verilator works internally.

Can you elaborate on what the issue is? I'm not sure I know what "initializing superclasses with local vars" is. The example you give above works with `-fno-localize` (which incorrectly localizes the funcout in the consgtructor call. Simple fix for that is don't localize stuff used inside an ExprStmt.

wsnyder, 2025-08-21:
I think that _constructor methods work, but as mentioned we don't need to do that now. So we probably should discuss the _constructor idea later in another bug unless @kbieganski you want to fix it that way now.  Just so it doesn't get lost though, I think the thing to do would be make some hand-written C++ that looks like the eventual output, including how the ClassRef reference counting class handles work, to make sure everything plays nicely together, including with being able to use the existing container types without fundamental changes.  I prefer we not drift too far from how "normal human C++" would be written, so we can continue to use inheritance, virtual functions, standard library and to generally avoid compiler problems.  E.g. we shouldn't go so far as to make our own `C` functions to replace `new`  (not that @gezalore is proposing that).

kbieganski, 2025-08-21:
Sorry, I missed that Geza mentioned our exact case. Thanks for the hint about `-flocalize`. We'll fix that.

igorosky, 2025-12-19:
@wsnyder could you give your thoughts on arguments mentioned here: https://github.com/verilator/verilator/issues/6828?

closed 2026-03-04 by wsnyder