verilator/verilator #6216: Simplify control flow statements

gezalore, 2025-07-22:
I'm working on more UNOPTFLAT fixes, but those require building a control flow graph for always blocks. AstWhile and AstJumpBlock are a bit of a pain. Can we simplify them along the following lines?

AstWhile: Remove `precondp`. This contains statements that need to be emitted both before the While itself, and after the 'incp' statements, at the end of the loop body. This is used in so few places, we could just clone the statements in `precondp` and insert them before the while, and append to 'incp'. Should yield the same C++ with some trivial Ast size increase, while enabling more straight forward flow analysis as a statement does not magically execute in multiple basic blocks.

AstJumpBlock: This is part of the AstJumpBlock/AstJumpGo/AstJumpLabel triplet. JumpGo and JumpLabel are both below the JumpBlock, and JumpGo references JumpLabel. However, JumpLabel is always at the very end of the JumpBlock. There are hits in the comments in  the definition of these nodes that "currently only forward jumping is allowed". However, this seems to have been enough for many years now. Can we therefor delete JumpLabel, and define that JumpGo branches to after the referenced (and enclosing, but not necessarily the nearest enclosing) JumpBlock? that makes it clearer how Jumps can be arranged when converting to a control flow graph.

wsnyder, 2025-07-23:
The uglyness with precondp is when we need to modify the expression for some pre-computation we'll need to clone, and the hard part is putting the statements in the "right" position.  Without precondp I don't think there's another correct place for them to go with "continue" statements still working correctly (we can't just add statements at end of the block), and we don't have another op free to have two copies, but maybe I'm missing something.

I think if we clean up JumpBlock what you are suggesting is it is a "real" basic-block.  I'd prefer we still have a way to optimize multiple jump blocks into one (in V3Const) but otherwise am fine if the change all works out.

gezalore, 2025-07-23:
> The uglyness with precondp is when we need to modify the expression for some pre-computation we'll need to clone

Yeah, cloning is fine, it's not a huge amount of trees.

> and the hard part is putting the statements in the "right" position. Without precondp I don't think there's another correct place for them to go with "continue" statements still working correctly [...]

I think it just works by putting it both before the loop, and append to `incsp`. Currently the While is emitted as such:

```c++
  iterateAndNextConstNull(nodep->precondsp());
  putns(nodep, "while (");
  iterateAndNextConstNull(nodep->condp());
  puts(") {\n");
  iterateAndNextConstNull(nodep->stmtsp());
// << 'conttnue goes here'
  iterateAndNextConstNull(nodep->incsp());
  iterateAndNextConstNull(nodep->precondsp());  // Need to recompute before next loop
  puts("}\n");
```

instead of

```
nodep->addPpecondsp(foo)
```

we can do:

```
nodep->addHereThisAsNesxt(foo->cloneTreePure(/* cloneNext: */ true));
nodep->addIncsp(foo);
```

I will go and try this out unless you see something broken with it.

> I think if we clean up JumpBlock what you are suggesting is it is a "real" basic-block.

Not quite. The body of the JumpBlock can be arbitrarily complex, as before.

> I'd prefer we still have a way to optimize multiple jump blocks into one (in V3Const)

What I'm proposing is equivalent to the current trifecta of Jump*, just with the extra unused expressivity removed, so anything that works now should work with the simplified version. Again I will have a go a this.

wsnyder, 2025-07-23:
>I think it just works by putting it both before the loop, and append to incsp.

I vaguely recall starting with that, but can't remember why I gave up and did precond.  A lot has changed since so maybe it will work now.  If you don't mind potentially having to back out if hit a problem, go for it.

Almost certainly already your plan, but separate pulls for each please.