verilator/verilator #6220: Skipping identical run overwrites stats file

gezalore, 2025-07-23:
When running verilator twice, the second run overwrites the stats file even if --skip-identical took effect (and then the contents of the new stats file are lacking).

Reproduce:

```
rm -rf test_regress/obj_vlt
test_regress/t/t_opt_ifjumpgo.py # Passes
test_regress/t/t_opt_ifjumpgo.py # Fails - stats file does not contain expected line
test_regress/t/t_opt_ifjumpgo.py --no-skip-identical # Passes again
```

wsnyder, 2025-07-23:
Um, this is intentional, it represents what the latest run did.  Does this cause problems outside tests?

gezalore, 2025-07-23:
Not causing problems, just a bit bizarre. Based on the docs one might expected obj_dir to not change at all if the run is actually skipped, but oh well. https://verilator.org/guide/latest/exe_verilator.html?highlight=skip%20identical#cmdoption-no-skip-identical

wsnyder, 2025-07-23:
Update the skip identical test to check it doesn't change.