verilator/verilator #6222: Regression, %Error: Internal Error: /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/rvlab_tap_pkg.sv:1:9: ../V3LinkCells.cpp:625: Module should be found globally if inserted in lib

Kreijstal, 2025-07-23:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

What 'verilator' command line do we use to run your example?

What 'verilator --version' are you using?  Did you try it with the git master version?
```
kreijstal@kreijstalnuc:~/git/risc-v-lab/build/sw_rlight/combined/sv_out$ verilator --version
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LC_CTYPE = "en_GB.UTF-8",
        LC_NUMERIC = (unset),
        LC_COLLATE = (unset),
        LC_TIME = (unset),
        LC_MESSAGES = (unset),
        LC_MONETARY = (unset),
        LC_ADDRESS = (unset),
        LC_IDENTIFICATION = (unset),
        LC_MEASUREMENT = (unset),
        LC_PAPER = (unset),
        LC_TELEPHONE = (unset),
        LC_NAME = (unset),
        LANG = "de_DE.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("de_DE.UTF-8").
Verilator 5.039 devel rev v5.038-60-g763183f06
```

What OS and distribution are you using?
archlinux
May we assist you in trying to fix this in Verilator yourself?
I have no idea why the error is there
(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

This was minimized with sv-bugpoint. Run with `--binary`
```

`line 429 "/home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/dm_pkg.sv" 0
`line 1 "/home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/rvlab_tap_pkg.sv" 1
package rvlab_tap_pkg;

`line 32 "/home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/rvlab_tap_pkg.sv" 0
endpackage

`line 27 "/home/kreijstal/git/risc-v-lab/src/rtl/rvlab_fpga/iocell_bidir.sv" 0
`line 1 "/home/kreijstal/git/risc-v-lab/src/rtl/rvlab_fpga/iocell_opendrain.sv" 1
module iocell_opendrain (
);
  IOBUF iocell (
    .O (in),   
    .IO(pad),   
    .I ('0),    
    .T (~oe)    
  );
endmodule

`line 429 "/home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/dm_pkg.sv" 0
`line 1 "/home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/rvlab_tap_pkg.sv" 1
package rvlab_tap_pkg;

`line 32 "/home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/rvlab_tap_pkg.sv" 0
endpackage

`line 85 "/home/kreijstal/git/risc-v-lab/build/vdeps/fetch/glbl.v" 0
```

wsnyder, 2025-07-23:
Note this already gets a MODDUP warning. I will make that into an error as it causes the later problem.

Kreijstal, 2025-07-23:
the thing is there is an external tool that calls the file like this
```
verilator -E -y /home/kreijstal/git/risc-v-lab/build/vdeps/fetch/verilator-unisims --Wno-fatal --Wno-EOFNEWLINE -DINIT_MEM_FILE=/home/kreijstal/git/risc-v-lab/build/sw_test_rvlab/build/sw.mem /home/kreijstal/git/risc-v-lab/src/rtl/inc/prim_assert.sv /home/kreijstal/git/risc-v-lab/src/rtl/rvlab_fpga/pkg/top_pkg.sv /home/kreijstal/git/risc-v-lab/src/rtl/prim/pkg/prim_pkg.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/pkg/ibex_pkg.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/pkg/tlul_pkg.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/dm_pkg.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/pkg/rvlab_tap_pkg.sv /home/kreijstal/git/risc-v-lab/build/xbar/generate/tl_peri_pkg.sv /home/kreijstal/git/risc-v-lab/build/xbar/generate/xbar_peri.sv /home/kreijstal/git/risc-v-lab/build/xbar/generate/tl_main_pkg.sv /home/kreijstal/git/risc-v-lab/build/xbar/generate/xbar_main.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_i2c_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_i2c_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_dma_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_dma_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/rv_timer_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/rv_timer_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_reverb_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_reverb_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_biquad_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_biquad_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_decider_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_decider_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/ddr_ctrl_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/ddr_ctrl_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/regdemo_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/regdemo_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/irq_ctrl_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/irq_ctrl_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/uart_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/uart_reg_top.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_rlight_reg_pkg.sv /home/kreijstal/git/risc-v-lab/build/reggen/generate/rtl/student_rlight_reg_top.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_timer/rv_timer.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_timer/timer_core.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_adapter_reg.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_assert.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_assert_multiple.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/sram2tlul.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_adapter_sram.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_err_resp.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_fifo_async.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_socket_m1.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_fifo_sync.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_socket_1n.sv /home/kreijstal/git/risc-v-lab/src/rtl/tlul/tlul_err.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/tlul_adapter_host.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/rvlab_tap.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/dm_mem.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/rv_dm.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/debug_rom.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/dmi_jtag.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/dm_sba.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/rvlab_debug.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/dm_csrs.sv /home/kreijstal/git/risc-v-lab/src/rtl/rv_dm/dmi_cdc.sv /home/kreijstal/git/risc-v-lab/src/rtl/student/student_rlight.sv /home/kreijstal/git/risc-v-lab/src/rtl/student/student_tlul_mux.sv /home/kreijstal/git/risc-v-lab/src/rtl/student/student_dma.sv /home/kreijstal/git/risc-v-lab/src/rtl/student/student.sv /home/kreijstal/git/risc-v-lab/src/rtl/student/student_irq_ctrl.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_multdiv_slow.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_register_file.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_pmp.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_controller.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_cs_registers.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_id_stage.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_alu.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_multdiv_fast.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_compressed_decoder.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_if_stage.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_core_tracing.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_ex_block.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_decoder.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_prefetch_buffer.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_core.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_fetch_fifo.sv /home/kreijstal/git/risc-v-lab/src/rtl/ibex/ibex_load_store_unit.sv /home/kreijstal/git/risc-v-lab/src/ [... cut ...] ab_tests.sv /home/kreijstal/git/risc-v-lab/src/fv/tlul_test_host.sv /home/kreijstal/git/risc-v-lab/src/fv/rvlab_board.sv /home/kreijstal/git/risc-v-lab/src/fv/rvlab_test_utils.sv /home/kreijstal/git/risc-v-lab/src/fv/jtag_master.sv /home/kreijstal/git/risc-v-lab/build/vdeps/fetch/glbl.v -I../../../src/fv/ -I../../../src/rtl/
```

this just works with other tools such as questa or vivado, I wonder if there is a way to not include the file twice (verilator seeing the file is already including it)?

wsnyder, 2025-07-23:
I checked and about half the tools give a warning that says they are using latest, and half error, so I think you're just lucky up till now ;)  Specifically IEEE 3.13 "it shall be illegal to redeclare a name already declared by a prior declaration."

Normally I declare packages with an include guard.

```
// package.svh
`ifndef PACKAGE_SVH  // guard
`define PACKAGE_SVH

 package Package; 
....

`endif  // guard
```

wsnyder, 2025-07-23:
If this is super-annoying I probably can make the error suppressible and have it discard the duplicate, but prefer not to given it is illegal.

Kreijstal, 2025-07-23:
even after removing the includes I get duplicate error, this didn't happen before why does verilator include it twice?

closed 2025-07-28 by wsnyder