verilator/verilator #6231: V3DfgDfgToAst.cpp:48: Incorrect width in AstNode created from DfgVertex REPLICATE: 1 vs 8

em2machine, 2025-07-26:
```
%Error: Internal Error: <snip>: ../V3DfgDfgToAst.cpp:48: Incorrect width in AstNode created from DfgVertex REPLICATE: 1 vs 8
  213 |             A: rslt = {  {4'8{|srcA[63:56]}},
```

When I change to the following it successfully builds.
`  213 |             A: rslt = {  {8{|srcA[63:56]}},`

Perhaps not a big deal, but wanted to open if useful.

```
verilator \
   --cc \
   --top aerial_wrap \
   -sv \
   -DVERILATOR=1 \
   -DCOMMON_CELLS_ASSERTS_OFF=1 \
   --build-jobs 8 \
   --trace \
   --trace-structs \
   --trace-max-array 16 \
   -output-split 15000 \
   --threads 4 \
   --no-assert \
   -O3 \
```

`Darwin easyemachine.local 24.5.0 Darwin Kernel Version 24.5.0: Tue Apr 22 19:53:27 PDT 2025; root:xnu-11417.121.6~2/RELEASE_ARM64_T6041 arm64`

Thanks,
-e-

wsnyder, 2025-07-26:
Is this using the current master?  If so can you post a complete small example please?

em2machine, 2025-07-26:
Sorry, yes, forgot to include:
`Verilator 5.039 devel rev v5.038-76-g20bd80698 (mod)`

Will hopefully get a reduced example yet today.

em2machine, 2025-07-27:
Changing to:
 
```
        OP_0: rslt = {  {8{|srcA[63:56]}},
                         {8{|srcA[55:48]}},
                         {8{|srcA[47:40]}},
                         {8{|srcA[39:32]}},
                         {8{|srcA[31:24]}},
                         {8{|srcA[23:16]}},
                         {8{|srcA[15: 8]}},
                         {8{|srcA[ 7: 0]}}  };
```

Results in:

```
emachine@easyemachine verilator % verilator --cc test2.sv
- V e r i l a t i o n   R e p o r t: Verilator 5.039 devel rev v5.038-76-g20bd80698 (mod)
- Verilator: Built from 0.025 MB sources in 3 modules, into 0.030 MB in 9 C++ files needing 0.000 MB
- Verilator: Walltime 0.009 s (elab=0.000, cvt=0.004, bld=0.000); cpu 0.007 s on 1 threads
```

Thanks.

PR #6233 (2025-07-27): Fix Replicate with unsigned count but MSB set
Correctly compute witdh of AstReplicate in its constructor when the count is unsigned but its MSB is set.

Fixes #6231

---

It's a bit suspect that `V3Number::isNegative()` returns true if the MSB is set even if `V3Number::isSigned()` is false, however there are a couple places where we check `isNegative()` without `isSigned()`, so instead of changing that I made the AstReplicate constructor do the same a V3Width. Might be worth revisiting though to make sure there are no similar bugs elsewhere.