verilator/verilator #6254: Bad assignment to triand

L3odr0id, 2025-08-02:
Compilation of the following SystemVerilog module throws several errors:
```SystemVerilog
module a (output triand logic b [3:3][1:2]);
  assign b = '{'{'b0,'bx}};
  assign b = '{'{'bx,'bx}};
endmodule: a
```

Command:
```
verilator --cc --exe ./testbench.cpp --timing -Wno-fatal +1800-2023ext+sv test.sv
```
Used [testbench](https://github.com/DepTyCheck/verilog-model/blob/master/.github/workflows/conf/verilator/testbench.cpp)

Output:
```
%Error: test.sv:2:12: AND unexpected in assignment to unpacked array
    2 |   assign b = '{'{'b0,'bx}};
      |            ^
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.037 for more assistance.
%Error: test.sv:2:12: Slices of arrays in assignments have different unpacked dimensions, 2 versus 1
    2 |   assign b = '{'{'b0,'bx}};
      |            ^
%Error: Internal Error: test.sv:2:14: ../V3Slice.cpp:316: Array initialization should have been removed earlier
    2 |   assign b = '{'{'b0,'bx}};
      |              ^~
                        ... This fatal error may be caused by the earlier error(s); resolve those first.
```

Expected behavior:
This module should be compiled. `IEEE 10.3.2 The continuous assignment statement` says

> Nets can be driven by multiple continuous assignments or by a mixture of primitive outputs, module outputs,
> and continuous assignments.

Also given literals have correct structure according to dimensions.

This issue occurs in `Verilator 5.037 devel rev v5.036-207-g5a6d5ed96` on ubuntu.

Found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck)

wsnyder, 2025-08-02:
Examples that are not "real code" related to tristates, and more so triand, are low priority to get fixed.  I would suggest avoid exploring tristates for bugs using automated tools, as there are likely many.  Instead it would be great if you could consider making a pull to fix this sort of issue as you find them.

wsnyder, 2026-10-18:
Welcome a PR but otherwise unlikely to be a priority to fix.

closed 2026-10-18 by wsnyder