verilator/verilator #6256: Interacting with submodules when hierarchically Verilating module

JiwooPark-arm, 2025-08-04:
When a big module with submodules gets hierarchically Verilated, where do the "interactions" with module go/how does that happen? e.g., driving signals into the smaller module, etc.

gezalore, 2025-08-05:
The separately verialted hierarchy blocks are compiled into a library (using the `--lib-create` optoin), and are re-integrated into the design via DPI. The latter is done by emitting a SystemVerilog wrapper for each hierarchy block, then including those in the final verilation of the original. You can see the SV wrapper in the output, e.g. of the test `test_regress/t/t_hier_block.py`.

See also https://verilator.org/guide/latest/verilating.html#hierarchical-verilation

Assume answered but feel free to follow up.

closed 2025-08-05 by gezalore

wsnyder, 2025-08-05:
The submodules under the hier module, depending on size, are typically flattened (or not) within the hier's code.  I would suggest it is likely more efficient for you to run an experiment with your code to see what is generated; you can look at the C++ and hier's Verilog wrapper that is generated.