verilator/verilator #6265: bitset _M_getdata private member error

paul-demo, 2025-08-05:
Any ideas on this? Looks like <bitset> issue compiling with clang/AL2/Graviton3/aarch64

```
/usr/lib/gcc/aarch64-redhat-linux/7/../../../../include/c++/7/bitset:1568:35: error: '_M_getdata' is a private member of 'std::__cxx1998::_Base_bitset<3>'
        return std::_Hash_impl::hash(__b._M_getdata(), __clength);
```

wsnyder, 2025-08-05:
BTW you can easily test this; clone this repo and change the .github actions to use ubuntu-24.04-arm.

paul-demo, 2025-08-05:
I'm trying Verilator out on this new platform -- it's using clang++ 11. But the "7" in there means a really outdated version of glibc?

```
glibc-2.26-64.amzn2.0.4.aarch64
```

paul-demo, 2025-08-05:
Ah I see that mean's it's using GCC7's C++ headers, which are indeed very old. I probably need to build libc++ for clang from source or locate it correctly.

closed 2025-08-05 by paul-demo