verilator/verilator #6279: support Combined wire declaration and continuous assignment with struct

qaztronic, 2025-08-10:
Thanks for taking the time to report this.

What would you like added/supported?

support Combined wire declaration and continuous assignment with struct

What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.038 2025-07-08 rev v5.038

Can you attach an example that runs on other simulators?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

`
typedef struct packed {
  logic [7:0] addr;
  logic [15:0] data;
  logic enable;
} my_struct_t;

my_struct_t source_struct = '{addr: 8'hFF, data: 16'hAAAA, enable: 1'b1};

// wire my_struct_t my_wire_struct = source_struct; // this doesn't work

// this works
wire my_struct_t my_wire_struct;
assign my_struct_t my_wire_struct = source_struct; 
`

wsnyder, 2025-08-10:
This works fine for me, even in 5.038, assuming a module around it.  Can you provide a complete example and the error?

qaztronic, 2025-08-15:
Tried to make a simpler example but couldn't reproduce the error. 

there is an error with this build: https://github.com/qaztronic/AXI4_LIB/commit/9f4ae62f047799f28ff7d864d8a4ce8997db3cec

the test bench is here: https://github.com/qaztronic/AXI4_LIB/tree/master/sim/tests/sc_axi4_lite_register_file

below is the diff to reproduce the error:

diff --git a/src/kit/axi4_bus_rd_fifo_if.sv b/src/kit/axi4_bus_rd_fifo_if.sv
index 22d7e64..aa09595 100644
--- a/src/kit/axi4_bus_rd_fifo_if.sv
+++ b/src/kit/axi4_bus_rd_fifo_if.sv
@@ -35,8 +35,8 @@ interface axi4_bus_rd_fifo_if #(axi4_lite_pkg::axi4_lite_cfg_t C='{default: 0})

   // --------------------------------------------------------------------
   wire [ C.A-1:0] araddr;
-  wire [AR_W-1:0] ar_flat_in;
-  assign ar_flat_in = araddr;
+  wire [AR_W-1:0] ar_flat_in = araddr;^M
+  // assign ar_flat_in = araddr;^M

   wire [AR_W-1:0] ar_flat_out;
   /* verilator lint_off UNUSEDSIGNAL */
@@ -51,12 +51,12 @@ interface axi4_bus_rd_fifo_if #(axi4_lite_pkg::axi4_lite_cfg_t C='{default: 0})
   /* verilator lint_on UNUSEDSIGNAL */
   /* verilator lint_on UNDRIVEN */
   axi4_lite_r_t r;
-  wire [R_W-1:0] r_flat_in;
-  assign r_flat_in = r;
+  wire [R_W-1:0] r_flat_in = r;^M
+  // assign r_flat_in = r;^M

   wire [R_W-1:0] r_flat_out;
-  axi4_lite_r_t _r;
-  assign _r = r_flat_out;
+  axi4_lite_r_t _r = r_flat_out;^M
+  // assign _r = r_flat_out;^M
   wire [(8*C.N)-1:0] _rdata = _r.data;
   wire [        1:0] _rresp = _r.resp;

Thank you for looking into this error.

wsnyder, 2025-08-16:
Can you please use sv-bugpoint or otherwise to reduce the test case?

Also you didn't indicate what the error you get is - chances are by just looking at the error you can make a tiny test case.

Thanks

wsnyder, 2025-09-05:
If can provide an example please reopen.