verilator/verilator #6281: Mailbox Type Mismatch Error with Parameterized Classes

philipaxer, 2025-08-11:
**Bug Description**
Verilator reports a mailbox type mismatch error when passing a parameterized mailbox to a constructor, even when the parameter types are identical. The error suggests that mailbox__TzX and mailbox__TzY are different types, but they should be the same type.

**Error Message**
```
%Error: tb.sv:45:15: Function Argument expects a CLASSREFDTYPE 'mailbox__Tz2', got CLASSREFDTYPE 'mailbox__Tz1'
                   : ... note: In instance 'tb_top'
   45 |         drv = new(mbx);   
      |               ^~~
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.039 for more assistance.
%Error: Exiting due to 1 error(s)
```

Verilator Version: latest main
Platform: Linux

Minimal Reproducible Example
```
// Base transaction class
class BaseTransaction;
    function new();
    endfunction
endclass

// Parameterized transaction class
class MyTransaction #(parameter int WIDTH = 32) extends BaseTransaction;
    bit [WIDTH-1:0] payload;
    function new(bit [WIDTH-1:0] payload = 0);
        super.new();
        this.payload = payload;
    endfunction
endclass

// Driver class expecting specific mailbox type
class MyDriver #(parameter int WIDTH = 32);
    mailbox #(MyTransaction#(WIDTH)) mbx;  // Expects specific parameterized type
    
    function new(mailbox#(MyTransaction#(WIDTH)) mbx);
        this.mbx = mbx;
    endfunction
    
    task run();
        MyTransaction#(WIDTH) tx;
        forever begin
            mbx.get(tx);
            $display("Driver got transaction with payload: %h", tx.payload);
        end
    endtask
endclass

// Environment class - this is where the error occurs
class Environment;
    MyDriver#(32) drv;
    mailbox #(MyTransaction#(32)) mbx;  // This creates one mailbox type
    
    function new();
    endfunction
    
    function void build();
        mbx = new();
        // ERROR: The mailbox type from the declaration doesn't match 
        // what the driver constructor expects due to parameter resolution
        drv = new(mbx);  // This line will cause the mailbox type mismatch error
    endfunction
    
    task run();
        fork
            drv.run();
        join_none
    endtask
endclass

// Top-level testbench module
module tb_top;
    Environment env;
    
    initial begin
        MyTransaction#(32) tx;

        env = new();
        env.build();
        env.run();
        
        // Send a transaction
        tx = new(32'hDEADBEEF);
        env.mbx.put(tx);
        
        #100;
        $finish;
    end
endmodule
```
**Steps to Reproduce**

Save the code above to a file called tb.sv
Run: `verilator --binary --timing tb.sv`
Observe the mailbox type mismatch error

**Expected Behavior**
The code should compile successfully since both mailbox declarations use MyTransaction#(32) and should resolve to the same type.

**Actual Behavior**
Verilator reports a type mismatch error suggesting that mailbox #(MyTransaction#(32)) declared in the Environment class is a different type than mailbox#(MyTransaction#(WIDTH)) where WIDTH=32 in the Driver constructor.

wsnyder, 2025-11-28:
Good debugging. Maybe make a similarValueTree which uses similar DType instead of dtype comparison, for AstConst uses opEq, then for everything else checks for == identity (ignoring dtype, using the existing per-node same() functions).

philipaxer, 2025-12-14:
What I conceptually don't understand in V3Param is this: For parameter value comparision, the valuep is used AS IS. The issue I see is that this is the value as given directly in the source code.

Case A:  `parameter real MYPARAM = 32;`
Then valuep is a CONST of int with 32 bit with value 32. 

Case B i.e. `parameter  int MYPARAM = 1.5;`
Here valuep is a CONST of real with 64bit with value 1.5. 

In both cases, the RHS value is used for any parameter resolution as is. That is not always correct, see example below:

```
class Class_A #(parameter int myparam = 1.5);
 
   function void  print();
    $display("myparam: %h", myparam);
  endfunction
endclass

module tb_top;
        
    initial begin 
      Class_A#(2.0) a;
      Class_A#(2) b;
      Class_A c;

      a = new();
      b = a;
      c = a;
        
    $finish;
    end
endmodule
```

The actual paramter is int. All values should evaluate to 2, after casting to int. So All objects should be of the same type and comparision in V3Param should be made on int granularity. That means values must be casted and then compared. This is what i see as the actual root cause - not the comparision per-se.

In my PR i tried to replace the RHS AstConst with a casted version. But this seems to break when the LHS type is more complicated, like below:

```
parameter integer MYPARAM = 12;
parameter logic[MYPARAM-1:0] MYPARAM2  = 0;
```

I just dont know if I am on the right track here...

regards
Philip