verilator/verilator #6301: CI: Add issue report on rtlmeter PR

wsnyder, 2025-08-17:
@gezalore but low priority,

Adding 'pr: rtlmeter' asks for the RTL meter tests to run, thanks for that.

Might that also compare the final results to the last night's run, and add a summary table to the issue?

Similar to performance table here https://github.com/verilator/verilator/pull/6168#issue-3218504901 though I'd like both verilation + simulation numbers.

gezalore, 2025-08-17:
> Also we should trigger and make the report on the 'labeled' workflow dispatch event, so we get both when the label is added after the pull was created.

Not sure If I'm misunderstanding. The workflow is already run on 'labeled' (and also on 'unlabeled', so removing the label will cancel the in progress run). I assume you just want a report when it's done.

gezalore, 2025-08-17:
Yes, tried it: #6271 (I always just add it later, not when opening the PR)

> I perhaps misassumed it wouldn't run on label changes because of no 'labeled' code present here:

The event_name is 'pull_request' (one of the keys under `on` below). The fact it's the 'labeled' "activity type" should be deducable somehow from `github.event`, but it doesn't matter here.
https://github.com/verilator/verilator/blob/dbdf23511578e4eca63604771d383a3f308ea465/.github/workflows/rtlmeter.yml#L8-L18
See also https://docs.github.com/en/actions/reference/workflows-and-actions/events-that-trigger-workflows#pull_request