verilator/verilator #6323: V3DfgSynthesize.cpp:442: Mismatched width reached DFG

L3odr0id, 2025-08-22:
Compiling the following SystemVerilog code produces multiple errors:
```SystemVerilog
module a(output logic [1:2] a1);
  assign a1 = 'bz;
endmodule: a

module b (output logic b1);
  a a_inst(.a1(b1));
endmodule: b
```

Command:
```
verilator --cc --exe ./testbench.cpp --timing -Wno-fatal --no-std +1800-2023ext+sv test.sv
```
Used [testbench](https://github.com/DepTyCheck/verilog-model/blob/master/.github/workflows/conf/verilator/testbench.cpp)

Output:
```
%Error-UNSUPPORTED: test.sv:7:13: Unsupported tristate port expression: VARREF '__Vcellinpt__a_inst__a1'
                               : ... note: In instance 'b'
    7 |   a a_inst(.a1(b1));
      |             ^~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.039
%Error: Internal Error: test.sv:7:13: ../V3DfgSynthesize.cpp:442: Mismatched width reached DFG
    7 |   a a_inst(.a1(b1));
      |             ^~
                        ... This fatal error may be caused by the earlier error(s); resolve those first.
```

Expected behavior:
This module should be compiled. The first error incorrectly says there is a tristate port, but only variable ports are declared. The second error mentions a width mismatch, but it seems a single element port and a packed array of length one should be compatible.

This issue occurs in `Verilator 5.039 devel rev v5.038-214-gf506aa878` on ubuntu.

Found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck)

gezalore, 2025-08-22:
@wsnyder In the Dfg synthesis patch I removed a check that would have made Dfg bail on this (if an Assign with mismatched widths on LHS/RHS is found), but I turned it into an assert.

Do you prefer I add that back and hide these quietly or leave it in and debug the fallout? Theoretically all assignments should have matching widths by the time they reach the optimizers.

wsnyder, 2025-08-22:
Dfg can/should assume assign widths match.

Once the first error occurs, anything else we can ignore.  In this case, as it indicates, we aren't supporting this sort of Z assignment with width mismatches, once we do support that the rest of the errors should either disappear or get fixed as part of that support.  Generally this sort of z expression isn't a high priority, so as always  a pull would be welcome if this is a priority for you.

gezalore, 2025-08-22:
Ah, I missed it was Error-UNSUPPORTED, makes sense then.

wsnyder, 2025-08-22:
We do exit just later, it's annoying to users not to see most errors in a single run.

wsnyder, 2026-10-17:
This properly gives an UNSUPPORTED message in master now.  Due to the complexities and uncommonness of such connections I don't see us planning to support this for the medium term, so closing.

closed 2026-10-17 by wsnyder