verilator/verilator #6341: Veer DFG compile error

wsnyder, 2025-08-28:
In verilator_ext_tests, running `t/t_cores_veer_eh1_cmark.py --vlt` fails with below compile error. adding `-fno-dfg` makes it pass.  I didn't bisect but it was passing on Sunday.

Likewise t_cores_swerv_cmark, but certainly same issue.

Attn @gezalore 

```
Vtb_top___024root__DepSet_h9ad02603__4.cpp: In function âvoid Vtb_top___024root___nba_comb__TOP__0(Vtb_top___024root*)â:
Vtb_top___024root__DepSet_h9ad02603__4.cpp:6612:62: error: array subscript 5 is outside array bounds of âVlWide<5> [1]â [-Werror=array-bounds=]
 6606 |                             ? 0U : (__Vtemp_4[((IData)(3U)
      |                                     ~~~~~~~~~~~~~~~~~~~~~~    
 6607 |                                                + (7U
      |                                                ~~~~~          
 6608 |                                                   & (((IData)(0x4eU)
      |                                                   ~~~~~~~~~~~~~~~~~~
 6609 |                                                       *
      |                                                       ~       
 6610 |                                                       (1U
      |                                                       ~~~     
 6611 |                                                        & (IData)(vlSelfRef.tb_top__DOT__rvtop__DOT__mem__DOT__iccm__DOT____Vcellout__genblk2__DOT__rd_addr_ff__dout)))
      |                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 6612 |                                                      >> 5U)))]
      |                                                      ~~~~~~~~^
Vtb_top___024root__DepSet_h9ad02603__4.cpp:6504:24: note: at offset 20 into object â__Vtemp_4â of size 20
 6504 |     VlWide<5>/*159:0*/ __Vtemp_4;
      |                        ^~~~~~~~~
```

gezalore, 2025-08-28:
Fails for me with f506aa878, from Friday. Note you need to configure with --enable-ccwarn, so maybe on Sunday you didn't but regardless, I will look into it.

wsnyder, 2025-08-28:
Oh, last Sunday I had a 22.04 system, now I'm 24.04, sorry didn't think of that obvious critical difference.  Appreciated if you can keep looking.

This also makes me realize I need to update the github cron for verilator_ext_tests to do 24.04 also, will do that once this is fixed.

wsnyder, 2025-08-28:
Also I thought Veer was part of rtlmeter, why not failing there?

gezalore, 2025-08-28:
No worries, I will track it down tomorrow.

gezalore, 2025-08-29:
This is pretty hairy.

The problem code is something like this, `x = y[idx*78 +: 78]` with idx being a 1 bit signal, and y 156 bits (twice 78), so x is either the low or high half of y. 

The result of the select is 3 words.

In V3Expand where we expand this assignment, when computing the top word of the result the code would pick up from `y` words "VL_BITWORD_E(idx*78) + 2" and "VL_BITWORD_E(idx*78) + 3" to then shift and or them together to extract the bits. The higher word can in fact be out of bounds for `y` when selecting the higher half (idx == 1).  However for a well formed select that is not out of bounds itself, we will never use the bits that come from the OOB access, as V3Clean has inserted an And mask over the result of the Sel, so the OOB bits in the high word of the Sel will be masked out so the execution is fine.

FWIW This has been the implementation since the 2006 root commit in git.

GCC has evolved the ability to sometimes figure out there is an OOB access because we use a conditional to detect an over-size shift (when the select is word aligned, the high word is not actually needed, but ShiftL by 32 is malformed), and that conditional sometimes have enough information to infer the variable idx is one, hence the warning.

I can't see a way to fix the OOB access without inserting another branch when computing the high word of the expanded Sel, which is unfortunate.

Using AstShiftLOvr instead of adding the conditional hides this from GCC so the warning goes away but that does sound like a bodge.

This also makes me realize asan is probably useless at runtime because of this.

This is the relavant code in V3Expand:
https://github.com/verilator/verilator/blob/8868d459a295b0b72bbc1925b7ae1ccaabca5295/src/V3Expand.cpp#L508-L532

gezalore, 2025-08-29:
BTW we emit a very high number of common sub-expression here. I will try to do some tidying. Let me know how you wan to tackle this (disable the warning, or add the extra branch for the high word). I'm tempted to bite the bullet and add the branch so asan can be useful.

wsnyder, 2025-08-29:
See #2819, which suggests adding a branch is not sufficient because in other cases the shift can "half-cross" boundaries.  I didn't come up with any efficient way around #2819 yet.

Assuming you agree with that, I'm ok with disabling the warning, as see no great alternative.

gezalore, 2025-08-29:
Not sure under what case the branch would be not sufficiet. When the Select itself is not OOB (that is, the user input SV is well formed). Then we only do an OOB read when computing the high word of the select, so instead of:

```
(from[idx + 2] >> 2) | (from[idx + 3] << 30)
```

do:

```
(from[idx + 2] >> 2) | ((idx + 3 == Const ? 0 : from[idx + 3]) << 30)
``

Where `Const` is the number of words i `from`. Would that not be enough? That's one extra branch for the high word only and would make non-OOB input produce non-OOB output. If the user's code is OOB, then fair game to blow up.

gezalore, 2025-08-29:
I have a patch where they now come out like this. I will run benchmarks overnight before posting it. There are at most 2 branches per expanded Sel and no OOB if the Sel itself is not OOB.

```c++
       __VExpandSel_Index_1 = (0xffU & ((IData)(0x4eU)`
                                        * (1U & (IData)(vlSelfRef.ifu_iccm_mem__DOT____Vcellout__genblk2__DOT__rd_addr_ff__dout))));
       __VExpandSel_WordIdx_1 = (__VExpandSel_Index_1`
                                 >> 5U);
       __VExpandSel_LoShift_1 = (0x1fU & __VExpandSel_Index_1);
       __VExpandSel_Aligned_1 = (0U == __VExpandSel_LoShift_1);
       if (__VExpandSel_Aligned_1) {
           __VExpandSel_HiShift_1 = 0U;
           __VExpandSel_HiMask_1 = 0U;
       } else {
           __VExpandSel_HiShift_1 = ((IData)(0x20U) - __VExpandSel_LoShift_1);
           __VExpandSel_HiMask_1 = 0xffffffffU;
       }
       __Vtemp_5[0U] = (((__Vtemp_4[((IData)(1U) + __VExpandSel_WordIdx_1)]`
                          << __VExpandSel_HiShift_1) & __VExpandSel_HiMask_1)`
                        | (__Vtemp_4[__VExpandSel_WordIdx_1]`
                           >> __VExpandSel_LoShift_1));
       __Vtemp_5[1U] = (((__Vtemp_4[((IData)(2U) + __VExpandSel_WordIdx_1)]`
                          << __VExpandSel_HiShift_1) & __VExpandSel_HiMask_1)`
                        | (__Vtemp_4[((IData)(1U) + __VExpandSel_WordIdx_1)]`
                           >> __VExpandSel_LoShift_1));
       __Vtemp_5[2U] = (((((2U <= __VExpandSel_WordIdx_1)
                            ? 0U : __Vtemp_4[((IData)(3U)`
                                              + __VExpandSel_WordIdx_1)])`
                          << __VExpandSel_HiShift_1) & __VExpandSel_HiMask_1)`
                        | (__Vtemp_4[((IData)(2U) + __VExpandSel_WordIdx_1)]`
                           >> __VExpandSel_LoShift_1));
```

PR #6345 (2025-08-30): Improve wide Sel expansion and substitution (#6341)
- Avoid repeated sub-expressions in expanded wide AstSel
- Fix OOB access to VlWide when the AstSel itself is not OOB

This also makes the output code a bit more readable and is overall neutral in performance for both verilation and simulation.

Fixes #6341
closed 2025-08-30 by wsnyder