verilator/verilator #6353: z3 runs for > hour for a simple test

solomatnikov, 2025-09-01:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

```
// Copyright (c) 2019 ETH Zurich, University of Bologna
//
// Copyright and related rights are licensed under the Solderpad Hardware
// License, Version 0.51 (the "License"); you may not use this file except in
// compliance with the License.  You may obtain a copy of the License at
// http://solderpad.org/licenses/SHL-0.51. Unless required by applicable law
// or agreed to in writing, software, hardware and materials distributed under
// this License is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
// CONDITIONS OF ANY KIND, either express or implied. See the License for the
// specific language governing permissions and limitations under the License.

// Package with functions and tasks commonly used in constrained randomized verification
package rand_verif_pkg;
  task automatic rand_wait(input int unsigned min, max, ref logic clk);
      int unsigned cycles = $urandom_range(max,min);
      repeat (cycles) @(posedge clk);
  endtask
endpackage
module clk_rst_gen #(
  parameter time          ClkPeriod = 0ps,  
  parameter int unsigned  RstClkCycles = 0
) (
  output logic clk_o,
  output logic rst_no
);
  logic clk;
  initial begin
    clk = 1'b0;
  end
  always begin
    clk = 1'b1;
    #(ClkPeriod / 2);
    clk = 1'b0;
    #((ClkPeriod + 1) / 2);
  end
  assign clk_o = clk;
  initial begin
    static int unsigned rst_cnt = 0;
    rst_no = 1'b0;
    #(ClkPeriod / 2);  
    while (rst_cnt < RstClkCycles) begin
      @(posedge clk);
      rst_cnt++;
    end
    rst_no = 1'b1;
  end
endmodule
module fifo_v3 #(
    parameter bit          FALL_THROUGH = 1'b0,  
    parameter int unsigned DATA_WIDTH   = 32,    
    parameter int unsigned DEPTH        = 8,     
    parameter type dtype                = logic [DATA_WIDTH-1:0],
    parameter int unsigned ADDR_DEPTH   = (DEPTH > 1) ? $clog2(DEPTH) : 1
)(
    input  logic  clk_i,             
    input  logic  rst_ni,            
    input  logic  flush_i,           
    input  logic  testmode_i,        
    output logic  full_o,            
    output logic  empty_o,           
    output logic  [ADDR_DEPTH-1:0] usage_o,   
    input  dtype  data_i,            
    input  logic  push_i,            
    output dtype  data_o,            
    input  logic  pop_i              
[... 279 lines cut ...]
    end
endmodule
```

What 'verilator' command line do we use to run your example?

```
verilator --sv --binary -Wno-fatal --trace-fst test.sv && cd obj_dir && time ../Vtest +wave
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
Verilator 5.040 2025-08-30 rev v5.040
```

What OS and distribution are you using?

```
Linux 5.14.0-570.37.1.el9_6.x86_64
```

May we assist you in trying to fix this in Verilator yourself?

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2025-09-14:
@solomatnikov note this is pending on you, or if you prefer please close, thanks.

wsnyder, 2026-10-26:
If you get it down to a very small test case please file a new issue. This test seems to fail for completely unrelated reasons and regardless I'm not going to debug or reduce large examples. Use claude or otherwise yourself to debug.

closed 2026-10-26 by wsnyder