verilator/verilator #6359: v5.040 failed to build on i386 arch

aelmahmoudy, 2025-09-03:
What 'verilator --version' are you using?  Did you try it with the git master version?
 verilator 5.040

What OS and distribution are you using?
Debian sid

verilator failed to build on i386 arch, below is the relevant build error:
```
ccache g++ -g -O2 -ffile-prefix-map=/builds/electronics-team/verilator/debian/output/source_dir=. -fstack-protector-strong -Wformat -Werror=format-security -g -O2 -ffile-prefix-map=/builds/electronics-team/verilator/debian/output/source_dir=. -fstack-protector-strong -Wformat -Werror=format-security  -Wdate-time -D_FORTIFY_SOURCE=2 -Wdate-time -D_FORTIFY_SOURCE=2  -O3 -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Wno-shadow -Wno-unused-parameter -faligned-new -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/share/verilator\" -include V3PchAstNoMT.h -c ../V3DfgCache.cpp -o V3DfgCache.o
In file included from ../V3DfgBreakCycles.cpp:19:
../V3Dfg.h: In instantiation of 'const T& DfgVertex::getUser() const [with T = long long unsigned int]':
../V3DfgBreakCycles.cpp:676:51:   required from here
  676 |         if (pair.second && vtxp->getUser<uint64_t>() == m_component) {
      |                            ~~~~~~~~~~~~~~~~~~~~~~~^~
../V3Dfg.h:841:29: error: static assertion failed: Size of user data type 'T' is too large for allocated storage
  841 |     static_assert(sizeof(T) <= sizeof(UserDataStorage),
      |                   ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
../V3Dfg.h:841:29: note: the comparison reduces to '(8 <= 4)'
../V3Dfg.h: In instantiation of 'void DfgVertex::setUser(T) [with T = long long unsigned int]':
../V3DfgBreakCycles.cpp:1103:40:   required from here
 1103 |                 selp->setUser<uint64_t>(vtxp->getUser<uint64_t>());
      |                 ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
../V3Dfg.h:856:29: error: static assertion failed: Size of user data type 'T' is too large for allocated storage
  856 |     static_assert(sizeof(T) <= sizeof(UserDataStorage),
      |                   ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
../V3Dfg.h:856:29: note: the comparison reduces to '(8 <= 4)'
make[4]: *** [../Makefile_obj:412: V3DfgBreakCycles.o] Error 1
```

Complete build log is at: https://salsa.debian.org/electronics-team/verilator/-/jobs/8173217/raw

gezalore, 2025-09-03:
Thanks, there was an assumption about pointers being 64-bit, but I think this was fixed on master by e63ed0a93173176d539a758eb86be57ffddc02f4. I unfortunately don't have a working 32-bit compiler on hand and we no longer test 32-bit platforms in the CI.

Ping @wsnyder to discuss what to do about 32-bit platform support.

wsnyder, 2025-09-03:
Um, but it isn't working correctly due to what you reported, or am I confused?

aelmahmoudy, 2025-09-04:
> Um, but it isn't working correctly due to what you reported, or am I confused?

But @gezalore fixed it in commit e63ed0a93173176d539a758eb86be57ffddc02f4

wsnyder, 2025-09-04:
Ah, missed that it got fixed.