verilator/verilator #6394: Fix DfgVertexVar::setHasRdXRefs crash on hier ref error

wsnyder, 2025-09-06:
From last night sv-tests/third_party/tools/icarus/ivtest/ivltests/sv_import_hier_fail1.v

```
package P;
  integer x;
endpackage

module M;
  import P::x;
  integer y;
  always_comb y = x;
endmodule

module test;

  M m ();

  initial begin
    integer y;
    y = m.x; // This should fail. Imported identifiers are not visible through
             // hierarchical names.
    $display("FAILED");
  end

endmodule
```

Gives

```
%Error: /svaha/wsnyder/SandBox/sv-tests/third_party/tools/icarus/ivtest/ivltests/sv_import_hier_fail1.v:19:11: Can't find definition of 'x' in dotted signal: 'm.x'
                                                                                                             : ... note: In instance 'test'
   19 |     y = m.x;
        ... Known scopes under 'x': <no instances found>
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
```

The error is fine, and perhaps the error handling needs to get fixed, but DFG which is giving the segfault needs to change to give some assert/internal error instead, as segfaults (correctly IMO) always fail sv-tests.

Likewise sv_import_hier_fail2.v sv_import_hier_fail3.v but presumably identical.

Thanks

wsnyder, 2025-09-06:
@gezalore

gezalore, 2025-09-07:
FWIW after fixing this in DFG it will give:

```
%Error: foo.sv:19:11: Can't find definition of 'x' in dotted signal: 'm.x'
                    : ... note: In instance 'test'
   19 |     y = m.x;
      |           ^
        ... Known scopes under 'x': <no instances found>
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.041 for more assistance.
%Error: Internal Error: foo.sv:19:11: ../V3Const.cpp:2830: Not linked
   19 |     y = m.x;
      |           ^
                        ... This fatal error may be caused by the earlier error(s); resolve those first.
```