verilator/verilator #6395: Trouble on the horizon for large design simulation in Verilator because of LLVM/clang long term strategy

taylor-bsg, 2025-09-06:
We are trying to build large designs with Verilator. GCC appears to be fairly hopeless for this. LLVM initially builds the large designs for a moderate number of "tiles" but when the design is scaled up, LLVM output silently crashes. We tracked this down to a problem with the number of constructor initializers in the Vtb__Syms.cpp file. When this reaches 64K, the output is no good. 

After a shit-ton of debugging, we found the culprit is in LLVM's clang/include/clang/AST/DeclBase.h file. It defines a constant NumCtorInitializers, which is the log of the number of constructor initializers that it will handle before silently corrupting itself.  It is currently set to 16 (i.e. 2^16=64K). But what is especially surprising is that this value was 21 only a few years ago, then went to 19, 18, 17. Basically, the LLVM front end maintainers are stealing bits away from this field over time.  So we are seeing an exponential decrease in the number of modules that Verilator can simulate over time. 

We went to the Clang Frontend forum on discourse.llvm.org, and asked about this (https://discourse.llvm.org/t/plans-for-cxxconstructordecl/88200) and discovered that their current roadmap is to decrease this down to 13. At that point Verilator will have gone from being able to support 2 million modules to only 8192.  Now I understand that this could also be fixed by hacking Verilator to pieces but it seems very strange for the world's leading edge compiler to be exponentially reducing the universe of programs it supports. 

It turns out there is a very simple fix for this in LLVM (adding one 32-bit field per constructor), but at least one LLVM maintainer (who works at Nvidia) on the discussion argued against the idea because it would use more memory.

I want to draw awareness of this to those who might find this concerning. For any that are more savvy about LLVM RFQ etc process, here is the link to the discussion:

https://discourse.llvm.org/t/plans-for-cxxconstructordecl/88200

taylor-bsg, 2025-09-06:
With @bdemsky (mostly him), we have a patch for LLVM (pasted into that discourse link) so hierarchical builds are not required. (Although perhaps there are other benefits to hierarchical mode?)

wsnyder, 2025-09-07:
Glad you have a patch, but most users in the future won't want/be willing to recompile LLVM.

If you have an e.g. system with a bunch of cores, it means you can compile a single core, then compile the system that uses that compiled core, and the resulting build time should be massively less as the core is only compiled once.  The runtime will likely suffer - we have work to do on that.

taylor-bsg, 2025-09-07:
Well I think there are a number of levels -- 0:no patch available, 1:patch available, 2:patch available via configuration option in LLVM, 3:functionality always available in LLVM.  Ideally we get to 3, but 3 >> 2 >> 1 >> 0.  If you are using LLVM for Verilator, I think there is a decent chance you built it yourself.

wsnyder, 2025-10-08:
Avoiding C++ initializers in the constructor, by e.g. removing them and instead in the constructor calling an `init` method would work for submodules but doesn't help e.g. SystemC which needs its constructors, and doesn't provide a general solution.  Also it means that constructors are still huge, which likely will hit other compiler bugs or at least performance issues, as we've had many other issues with large structures and such.

Instead I'd suggest that whenever we have a class we are to emit with more than some programmable threshold of members, we instead move all of those members to a new class and instantiate that.   E.g. if we would previously had 1M members, and threshold was 1K, we would instead instantiate 1K new container classes each with 1K members each.

This is nice as would also fix the case if we ever get a Verilog class with too many members for the compiler.

For starters do this in a new pass affecting only the symbol table, which probably means mostly VSelfPointerText cleanups.

@taylor-bsg is this a pull you might be interested in contributing?

wsnyder, 2025-11-06:
I suggest generate a little standalone C++ program first to see what helps.  My quick reading was changing init style would not be sufficient, but would be happy to be wrong.

gezalore, 2025-11-06:
Sorry crucial omission on my part. The issue is the sheer number of them, not the style per se. 

However, there is no C++ syntax to split this (which is just 10000 function calls):

````
Foo::Foo()
  : mod0{this}
  , mod1{this}
  ...
  , mod10000{this}
{}
```

But we can split this into multiple smaller sub-functions:

````
Foo::Foo()
  mod0.configure(this);
  mod1.configure(this);
  ...
  mod10000.configure(this);
{}
```

Splitting them will be a follow up patch after the style change. Some compilers choke on too many function calls even with optimizations disabled.

wsnyder, 2025-11-23:
Assuming the pulls made this much better, if not please reopen or file a new issue, thanks.

closed 2025-11-23 by wsnyder