verilator/verilator #6431: Verilator internal fault

solomatnikov, 2025-09-13:
Thanks for taking the time to report this.

Can you please attach an example that shows the issue or missing feature?  (Must be openly licensed, completely self-contained so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs)

```
package uvm_pkg;                
virtual class uvm_sequence #(type REQ = int);
endclass
endpackage
package s_c_sequences_pkg;
  import uvm_pkg::*;
class p_reg_api extends uvm_sequence  ;
  function new(string name );
    super.new(name);
  endfunction
endclass
endpackage
module tb_top;
endmodule : tb_top
```

What output from that test indicates it is wrong, and what is the correct or expected output?  (Or, please make test self-checking if possible.)

What 'verilator' command line do we use to run your example?

```
verilator --binary --sv sv-bugpoint-minimized.sv
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
```

```
verilator --binary --sv --debug --gdbbt sv-bugpoint-minimized.sv
export VERILATOR_ROOT='~/.local/verilator/v5.041-gf7ad54b32/share/verilator'
No stack.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Starting Verilator 5.041 devel rev v5.040-107-gf7ad54b32
- Verilator.cpp:663:  Option --verilate: Start Verilation
- V3File.cpp:229:        --check-times failed: no input obj_dir/Vsv__02dbugpoint__02dminimized__verFiles.dat
- V3ParseImp.cpp:302: parseFile: verilated_std_waiver
- V3PreShell.cpp:90:  Preprocessing ~/.local/verilator/v5.041-gf7ad54b32/share/verilator/include/verilated_std_waiver.vlt
- V3PreShell.cpp:143:     Reading ~/.local/verilator/v5.041-gf7ad54b32/share/verilator/include/verilated_std_waiver.vlt
- V3ParseImp.cpp:394: Lexing ~/.local/verilator/v5.041-gf7ad54b32/share/verilator/include/verilated_std_waiver.vlt
- V3ParseImp.cpp:302: parseFile: verilated_std
- V3PreShell.cpp:90:  Preprocessing ~/.local/verilator/v5.041-gf7ad54b32/share/verilator/include/verilated_std.sv
- V3PreShell.cpp:143:     Reading ~/.local/verilator/v5.041-gf7ad54b32/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:394: Lexing ~/.local/verilator/v5.041-gf7ad54b32/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:302: parseFile: sv-bugpoint-minimized
- V3PreShell.cpp:90:  Preprocessing sv-bugpoint-minimized.sv
- V3PreShell.cpp:143:     Reading sv-bugpoint-minimized.sv
- V3ParseImp.cpp:371: Writing all preprocessed output to obj_dir/Vsv__02dbugpoint__02dminimized__inputs.vpp
- V3ParseImp.cpp:394: Lexing sv-bugpoint-minimized.sv
- V3LinkLevel.cpp:42: modSortByLevel()
- V3Ast.cpp:1377:     Dumping obj_dir/Vsv__02dbugpoint__02dminimized_002_cellsort.tree
- V3LinkDot.cpp:5035: linkDotPrimary:
- V3LinkJump.cpp:480: linkJump:
- V3Ast.cpp:1377:     Dumping obj_dir/Vsv__02dbugpoint__02dminimized_007_linkjump.tree
- V3LinkInc.cpp:350:  linkIncrements:
- V3Ast.cpp:1377:     Dumping obj_dir/Vsv__02dbugpoint__02dminimized_008_linkinc.tree
- V3Param.cpp:1737:   param:
- V3Ast.cpp:1377:     Dumping obj_dir/Vsv__02dbugpoint__02dminimized_009_param.tree
%Error: Internal Error: sv-bugpoint-minimized.sv:10:11: ../V3Broken.cpp:168: Broken link in node (or something without maybePointedTo): 'm_taskp && !m_taskp->brokeExists()' @ ./V3Ast__gen_impl.h:4784
                                                      : ... note: In instance 'tb_top'
-node: NEW 0x11110d0 <e1359> {e10al} @dt=0@  new
   10 |     [super.new](http://super.new/)(name);
      |           ^~~
                        ... See the manual at https://verilator.org/verilator_doc.html?v=5.041 for more assistance.
- V3Ast.cpp:1377:     Dumping obj_dir/Vsv__02dbugpoint__02dminimized_990_final.tree
- V3StatsReport.cpp:233:statsReport:
%Error: Internal Error: Aborting since under --debug
[... 57 lines cut ...]
#54 BrokenCheckVisitor::BrokenCheckVisitor (this=0x7fffffffb850, nodep=0x10b6b80) at ../V3Broken.cpp:327
#55 0x00000000005a4bd6 in V3Broken::brokenAll (nodep=0x10b6b80) at ../V3Broken.cpp:357
#56 0x0000000000619ba9 in V3Global::dumpCheckGlobalTree (stagename=..., newNumber=newNumber@entry=0, doDump=doDump@entry=true) at ../V3Global.h:154
#57 0x00000000008c6927 in V3Param::param (rootp=<optimized out>) at /opt/rh/gcc-toolset-12/root/usr/include/c++/12/bits/new_allocator.h:80
#58 0x000000000052f5e2 in process () at ../V3Global.h:154
#59 0x0000000000530e32 in verilate (argString=...) at ../Verilator.cpp:721
#60 0x000000000053206b in main (argc=<optimized out>, argv=0x7fffffffc688) at ../Verilator.cpp:878
A debugging session is active.
```

What 'verilator --version' are you using?  Did you try it with the git master version?  Did you try it with other simulators?

```
Verilator 5.041 devel rev v5.040-107-gf7ad54b32
```

What OS and distribution are you using?

```
Linux 4.18.0-553.22.1.el8_10.x86_64
```

May we assist you in trying to fix this in Verilator yourself?

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2025-09-13:
There's no `name` in the super's class, so this isn't legal code, probably because it over-minimized, please check examples with other simulators to make sure stays legal after minimization, thanks.

Anyhow I assumed this:

```
package uvm_pkg;
  virtual class uvm_sequence #(
      type REQ = int
  );
    function new(string name);
    endfunction
  endclass
endpackage

package seq_pkg;
  import uvm_pkg::*;
  class p_reg_api extends uvm_sequence #(int);
    class p_reg_api extends uvm_sequence;
    function new(string name);
      super.new(name);
    endfunction
  endclass
endpackage

module t;
endmodule
```

Problem's workaround is to pass default parameter to the base class:

```
class p_reg_api extends uvm_sequence #(int);
```

Presumably V3LinkDot/V3Param isn't realizing that it needs to defer processing the link until V3Param.

@ArturBieniek4 your recent work here was great, any interest in making a pull to fix this, as in a similar area?

wsnyder, 2025-09-13:
The internal error prevents a later error message, which means that bugpoint over-optimized (as it didn't know better) and ended up with something too small and illegal. My point was only is it's helpful in the future if you could please check the final minimized examples against other simulators before posting (given you file a lot).  All set on this one, your correction is similar to what I posted.

closed 2025-09-17 by wsnyder