verilator/verilator #6450: Unique If reports violation when it should not

deo248, 2025-09-17:
The following code reports a unique if violation when I believe it should not due to the "else null" at the end of the unique if chain.
```verilog
module tb;
    logic clk;
    logic A, B, C;
    logic reset;

    always #1 clk = ~clk;

    initial begin
        clk = 0;
        reset = 0;
        A = 0;
        B = 0;
        C = 0;
        #10;
        reset = 1;
        #2;
        A = 1;
        #10;
        $finish;
    end

    logic x;

    blah blah_i (
        .*
    );

endmodule

module blah (
    input logic reset,
    input logic clk,
    input logic A, B, C,
    output logic x
);

    always_ff @(posedge clk or negedge reset) begin
        if(!reset) begin
            x <= '0;
        end else if(A) begin
            //this would really be best as a unique0, however we've found in practice
            //that unique is fairly well-supported across a range of tools while
            //unique0 is not
            unique if(B) begin
                x <= '0;
            end else if (C) begin
                x <= '1;
            end else ;//necessary for unique if to not have a false positive
        end
    end

endmodule
```
The following functionally-equivalent module does not issue a unique if violation and uses a different style of else condition.
```verilog
    always_ff @(posedge clk or negedge reset) begin
        if(!reset) begin
            x <= '0;
        end else if(A) begin
            //this would really be best as a unique0, however we've found in practice
            //that unique is fairly well-supported across a range of tools while
            //unique0 is not
            unique if(B) begin
                x <= '0;
            end else if (C) begin
                x <= '1;
            end else begin
                ;//necessary for unique if to not have a false positive
            end
        end
    end
```

What output from that test indicates it is wrong, and what is the correct or expected output?
```
[13] %Error: unique_if_else.sv:44: Assertion failed in tb.blah_i: 'unique if' statement violated
%Error: unique_if_else.sv:44: Verilog $stop
Aborting...
```
Expected behavior: runs without error.

What 'verilator' command line do we use to run your example?
`verilator --timing --binary unique_if_else.sv`

What 'verilator --version' are you using?
5.040

Did you try it with the git master version?
No, can if desired.

Did you try it with other simulators?
No, can if desired.

What OS and distribution are you using?
RHEL9.

May we assist you in trying to fix this in Verilator yourself?
No, sorry.

wsnyder, 2025-09-17:
It had treated the `;` to mean no else, easy fix.

closed 2025-09-17 by wsnyder

deo248, 2025-09-17:
Appreciate the quick fix!