verilator/verilator #6456: `localparam` evaluated based on default module parameters when read from hierarchical path

LucaRufer, 2025-09-19:
When a `localparam` value of a module is _read_ from outside the module using a hierarchical path, the value of the `localparam` is computed based on the default parameters of the module, even if they are overwritten. For example, running the following code snippet:

```
module foo #(
  parameter int X = 1
) ();

  localparam int Y = X;
  localparam int Z = X;

  initial $display("X=%0d, Y=%0d, Z=%0d.", X, Y, Z);

endmodule

module tb;
  localparam int MY_X = 2;

  foo #(.X(MY_X)) inst ();

  localparam int MY_Y = inst.Y;

  initial $display("MY_Y=%0d.", MY_Y);

endmodule
```

prints the following lines:

```
MY_Y=1.
X=2, Y=1, Z=2.
```

Even tough all values are supposed to be `2`. Reading the value of `inst.Y` in the line `localparam int MY_Y = inst.Y;` causes the value to be wrong both inside and outside the module. The value of `inst.Z` is not affected and shows the correct value derived from the overwritten value of parameter `X`. The problem seems to originate in the constant propagation, where the value of `MY_Y` is evaluated before the module `inst` is evaluated.

The example snippet above was run with `Verilator 5.036 2025-04-27 rev v5.036`, running the command `verilator --debug --timing --cc --exe --main --build --top-module tb test.sv -o test.vlt` on `AlmaLinux 9.6`.

The example snippet was also tested with VCS, which produces the correct output:
```
MY_Y=2.
X=2, Y=2, Z=2.
```

The following snippet could be used in regession tests:
module foo #(
  parameter int X = 1
) ();
  localparam int Y = X;
endmodule

module tb;
  localparam int MY_X = 2;

  foo #(.X(MY_X)) inst ();

  localparam int MY_Y = inst.Y;

  initial begin
    if (MY_X == MY_Y) $finish; else $error;
  end
endmodule

LucaRufer, 2025-09-19:
Looking a bit further into this, I came across the following paragraph in the System Verilog language spec (1800-2023, Section 6.20.2):

> A value parameter (parameter, localparam, or specparam) can only be set to an expression of literals,
value parameters or local parameters, genvars, enumerated names, or a constant function of these. Package
references are allowed. Hierarchical names are not allowed.

Thus, I think the problem is not how Verilator evaluates the `localparam`, but rather that a localparam set to a value from a hierarchical path, which is not valid according to the spec. Still, I think a Warning (or even an Error) when attempting to set a (local)param from a hierarcical path would be nice.

wsnyder, 2025-09-20:
I understand some simulators take this, but others do not, and IEEE clearly says this is illegal:

   6.20.2 Value parameters    "...A value parameter (parameter, localparam, or specparam) can only be set to an expression of literals, value parameters or local parameters, genvars, enumerated names, or a constant function of these. Package     references are allowed. Hierarchical names are not allowed.

Which makes sense given how hard this will be.

Converting this issue to flag the error.