verilator/verilator #6486: C++ error: constructors cannot be declared ‘virtual’

solomatnikov, 2025-09-25:
Thanks for taking the time to report this.

Can you please attach an example that shows the issue or missing feature?  (Must be openly licensed, completely self-contained so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs)

```
class uvm_object;
    virtual function new(string name = "uvm_object");
    endfunction
endclass
class s_t_a_cfg extends uvm_object;
    extern function new(string name = "s_t_a_cfg");
endclass
function s_t_a_cfg::new(string name = "s_t_a_cfg");
    super.new(name);
endfunction

module t;
  initial begin
    $write("*-* All Finished *-*\n");
    $finish;
  end
endmodule
```

What output from that test indicates it is wrong, and what is the correct or expected output?  (Or, please make test self-checking if possible.)

```
In file included from Vtest__Syms.h:19,
                 from Vtest__pch.h:23,
                 from Vtest.cpp:4,
                 from Vtest__ALL.cpp:3:
Vtest___024unit__03a__03auvm_object__Vclpkg.h:36:5: error: constructors cannot be declared ‘virtual’ [-fpermissive]
   36 |     virtual Vtest___024unit__03a__03auvm_object(Vtest__Syms* __restrict vlSymsp, std::string name);
      |     ^~~~~~~
```

What 'verilator' command line do we use to run your example?

```
verilator --binary test.sv
```

What 'verilator --version' are you using?  Did you try it with the git master version?  Did you try it with other simulators?

```
Verilator 5.041 devel rev v5.040-196-g63f5f5c32
```

What OS and distribution are you using?

```
Linux  4.18.0-553.22.1.el8_10.x86_64
```

May we assist you in trying to fix this in Verilator yourself?

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2025-09-25:
@solomatnikov,

I appreciate the bugs you file have improved the quality of Verilator.  And you have nicely tuned the bug reports to my liking.

Unfortunately, I like to close bugs, and with your filing a bug nearly every day for weeks, and having nearly 20 bugs open, I find myself approaching burnout. Also, it distracts from working on more perhaps "community popular" backlog issues like general UVM. (Note Verilator takes my outside-work time; "is not my real job").

I therefore request you discontinue filing bugs that appear to be from your "searching" for issues and/or pushing to UVM which is still not supported.  At least limit to say one every few weeks.

Alternatively, and better, if you switch your (appreciated) time investments in testing to providing pull requests that show a problem, and also fix the Verilator code, that would be by far best as the issues would get fixed.

Alternative 2, please contract with a support organization to make pulls for the improvements you/your org desires.

Of course,  if there is an occasional bug that is truly blocking your work, please file it immediately.

Thanks for understanding.

P.S. I believe in full visibility on community projects, so posted this here, but if you prefer to discuss by private email, reach out.

closed 2025-09-27 by wsnyder

solomatnikov, 2025-09-29:
Sorry, if I did something wrong, it was certainly not the intent.

I do not file bugs by "searching" for issues and/or pushing to UVM. This particular issue came up when I was trying to create a small test case for #6498. Unfortunately, https://github.com/antmicro/sv-bugpoint gets stuck in some weird state, sometimes after running for days. When trying to manually edit as a workaround I hit this.

wsnyder, 2025-10-02:
There is a real carrying cost to keeping bugs open as they need to be checked periodically, probably totaling around an hour per open bug per year, so this takes time away from fixing issues.

The issues closed were thought to now be working, no longer applicable/reproducible, or some feature that no one is likely to work on in a reasonable timeframe of multiple years.    If there is a bug describing something that e.g. IEEE requires that was closed in error please comment in that bug.  As to more "feature-like" issues, as always, if there's something important to you, provide a pull or sponsor someone to make a pull.