verilator/verilator #6488: Configure script incorrectly detects no C++14 support

Steampunkery, 2025-09-25:
# System Information
OS: Ubuntu 24.04
Verilator: v5.040
gcc/g++: 12

# Bug
I am attempting to follow the [Git Quick Install](https://verilator.org/guide/latest/install.html#git-quick-install) guide. On my system, the `ld` linker emits a warning about having a different version of libjansson than it expects (this is normal and does not cause issues). However, due to the way that the configure script checks for C++14 support, if any line at all is present in the `conftest.err` file (such as a linker warning), autoconf will interpret this as a failure even though the return code is 0.

I have traced the issue to this autoconf macro: https://github.com/verilator/verilator/blob/v5.040/configure.ac#L289

The instantiation of this macro eventually leads to the following lines being present in the configure script:
```
if ac_fn_cxx_try_link "$LINENO"
then :
  _my_result=yes
      if test -s conftest.err; then
        if grep -e "" conftest.err >/dev/null; then
          _my_result=no
        fi
      fi
else $as_nop
  _my_result=no
fi
```

Note the empty string in the grep invocation. I got past this and was able to build verilator by just editing the script manually to not perform this check because I know that my version of g++ is compatible.

I am not sure what the proper way to fix this is because I am not well versed in autoconf.

wsnyder, 2025-09-26:
Try this:

```
diff --git a/configure.ac b/configure.ac
--- a/configure.ac
+++ b/configure.ac
@@ -590,18 +590,17 @@ AC_SUBST(CFG_LIBS)
 AC_DEFUN([_MY_CXX_CHECK_CXX_VER],
    [# _MY_CXX_CHECK_CXX_VER(flag) -- Check if compiler runs C++14
     # Set $_my_result
-    AC_LINK_IFELSE(
+    AC_RUN_IFELSE(
      [AC_LANG_PROGRAM([#include <thread>
 #if (__cplusplus < 201402L)
 # error "Too old"
 #endif
-    ], [[ ]])],
-     [_my_result=yes
-      if test -s conftest.err; then
-        if grep -e "$1" conftest.err >/dev/null; then
-          _my_result=no
-        fi
-      fi],
+    ], [[
+       std::thread t([]{});
+       t.join();
+     ]])],
+     [_my_result=yes],
+     [_my_result=no],
      [_my_result=no])
    ])

```

Steampunkery, 2025-09-26:
That patch solves the issue for me!

closed 2025-09-26 by wsnyder