verilator/verilator #6500: Verilator recognizes wire delay as a function error

elena-chernykh, 2025-09-29:
Hello!

I encountered a problem during CVA6 core simulation with Verilator (5.041 devel rev v5.040-217-g98786e609). The error is:
```
%Error: /home/username/cva6/corev_apu/tb/common/SimDTM.sv:41:15: Delays are not legal in functions. Suggest use a task (IEEE 1800-2023 13.4.4)
                                                                                     : ... note: In instance 'ariane_testharness'
   41 |   wire [31:0] #0.1 __debug_resp_bits_data = debug_resp_bits_data;
      |               ^
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.041 for more assistance.
```

I tried simulations with different Verilator releases (version 5.008 as required by CVA6, and the latest release 5.040), but this error only appears in version 5.041. Additionally, the error description "Delays are not legal in functions" appears to be misleading.

After modifying the SystemVerilog file SimDTM.sv as follows:
```
wire [31:0] __debug_resp_bits_data;
assign #0.1 __debug_resp_bits_data = debug_resp_bits_data;
```
the problem disappears. However, I would prefer not to modify the CVA6 source files directly.

Could you please take a look at the error?

gezalore, 2025-09-30:
Nice example, thank you. I suspect this is due to the recent memory leak fixes in the parser. I will look into it.

wsnyder, 2025-09-30:
@gezalore I recently added the error so if you prefer I look at it tonight let me know.